enjoy-digital/litex_m2sdr #28: SoapySDR improvements.

enjoy-digital, 2024-09-04:
Issues seen with GQRX:
- [ ] In Configure I/O devices, report a valid list of Input rates (for now user can configure it, but supported rates are not listed).
- [ ] PGA can only be controlled after Hardware AGC has been enabled/disabled.

trabucayre, 2024-09-04:
- for Sampling rate: GQRX has a list of hardcoded values [here](https://github.com/gqrx-sdr/gqrx/blob/master/src/qtgui/ioconfig.cpp#L227). Informations are not fetched from the driver

enjoy-digital, 2024-09-05:
@trabucayre: For selecting RX1/TX1 or RX2/TX2, it will in fact be easier to just switch to 1T1R mode (https://github.com/enjoy-digital/litex_m2sdr/issues/29) and configure the correct channel in 1T1R directly on the AD9361 RFIC, should be here: https://github.com/enjoy-digital/litex_m2sdr/blob/main/software/user/libm2sdr/m2sdr_config.h#L254-L255

enjoy-digital, 2025-03-06:
We can close this issue now since improvements have been implemented. If limitation, we could open other specific issues.

closed 2025-03-06 by enjoy-digital