enjoy-digital/litex_m2sdr #31: about OrangePi 5 Max PCIe setting

akioolin, 2024-10-02:
Hello All,

My current configuration is LiteX-M2SDR board + OrangePi 5 MAX.

I found the LiteX-M2SDR's LED is on and blinking a while during boot, but become blank after shell prompt appear.

Using lspci to check PCIe devices, but there is no detection of LiteX-M2SDR board. 

After tracing the code of LiteX-M2SDR, the PCIe configuration seems using 4Lanes. Am I wrong?

For RPi 5 / CM4 , PCIe is Gen2, for RK3588 seems Gen3 on M.2 M-key slot.

What is my missing point ?

Many Thanks.

BR, Akio

enjoy-digital, 2024-10-02:
Hi @akioolin,

the board you received should indeed be flashed with the 4-lanes bitstream. If possible, I would recommend doing a first test in classic computer with a M2 X4 connector just to verify enumeration with lspci or not. If enumerating, the issue will probably be related to OrangePi5 PCIe configuration. If not enumerating, I would recommend re-flashing the bitstream over JTAG with the M2 variant bitstreams available here:  https://github.com/enjoy-digital/litex_m2sdr/issues/30

@meriac: Do you remember if anything specific was required on the OrangePI 5 Max to get the PCIe M2 connrector up?

Florent

akioolin, 2024-10-02:
Hi, @enjoy-digital :

That is a very good idea. After testing, I'll report the result.

BR, Akio

akioolin, 2024-10-09:
Hi, @enjoy-digital , All: 
 
 After some study and test, below is relevant updates need to be made to the Device Tree settings of 
 OrangePi 5 Max. Please proceed in the following order.

 cd /boot/dtb/rockchip
 sudo cp rk3588-orangepi-5-max.dtb rk3588-orangepi-5-max.orig.dtb
 sudo dtc rk3588-orangepi-5-max.dtb -o rk3588-orangepi-5-max.dts
 sudo joe rk3588-orangepi-5-max.dts
 Find num-lanes in the pcie@fe150000 section, change its setting value from <0x04> to <0x02>, and save it.
 sudo dtc rk3588-orangepi-5-max.dts -o rk3588-orangepi-5-max.dtb

 After completion, reboot.

 sync;sync;sync;sudo reboot

 After logging into the system, confirm the following items to confirm whether the modification is successful.

 dmesg |grep fe150000, confirm that the following message appears.

 [6.267234] rk-pcie fe150000.pcie: invalid prsnt-gpios property in node
 [ 6.283941] rk-pcie fe150000.pcie: host bridge /pcie@fe150000 ranges:
 [ 6.283978] rk-pcie fe150000.pcie: IO 0x00f0100000..0x00f01fffff -> 0x00f0100000
 [ 6.284028] rk-pcie fe150000.pcie: MEM 0x00f0200000..0x00f0ffffff -> 0x00f0200000
 [ 6.284054] rk-pcie fe150000.pcie: MEM 0x0900000000..0x093fffffff -> 0x0900000000
 [ 6.284115] rk-pcie fe150000.pcie: iATU unroll: enabled
 [ 6.284122] rk-pcie fe150000.pcie: iATU regions: 8 ob, 8 ib, align 64K, limit 8G
 [ 6.489335] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x3
 [ 6.514894] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x4
 [ 6.598237] rk-pcie fe150000.pcie: PCIe Link up, LTSSM is 0x130011
 [ 6.598246] rk-pcie fe150000.pcie: PCIe Gen.2 x1 link up
 [ 6.598318] rk-pcie fe150000.pcie: PCI host bridge to bus 0000:00

 dmesg |grep "pci 0000:00"

 [6.598362] pci 0000:00:00.0: [1d87:3588] type 01 class 0x060400
 [6.598375] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x3fffffff]
 [6.598383] pci 0000:00:00.0: reg 0x14: [mem 0x00000000-0x3fffffff]
 [6.598390] pci 0000:00:00.0: reg 0x38: [mem 0x00000000-0x0000ffff pref]
 [6.598432] pci 0000:00:00.0: supports D1 D2
 [6.598436] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
 [6.605053] pci 0000:00:00.0: BAR 0: no space for [mem size 0x40000000]
 [6.605060] pci 0000:00:00.0: BAR 0: failed to assign [mem size 0x40000000]
 [6.605066] pci 0000:00:00.0: BAR 1: no space for [mem size 0x40000000]
 [6.605071] pci 0000:00:00.0: BAR 1: failed to assign [mem size 0x40000000]
 [6.605076] pci 0000:00:00.0: BAR 8: assigned [mem 0xf0200000-0xf02fffff]
 [6.605081] pci 0000:00:00.0: BAR 6: assigned [mem 0xf0300000-0xf030ffff pref]
 [6.605103] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
 [6.605110] pci 0000:00:00.0: bridge window [mem 0xf0200000-0xf02fffff]

 lspci, confirm that the following message appears.
 0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd Device 3588 (rev 01)
 0000:01:00.0 RF controller: Xilinx Corporation Device 7024
 0003:30:00.0 PCI bridge: Rockchip Electronics Co., Ltd Device 3588 (rev 01)
 0003:31:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
 If "RF controller: Xilinx Corporation Device 7024" appears, the system has detected the LiteX-M2SDR board.

 When all three of the above messages appear, you can proceed to the testing phase.

 Testing stage:

 Carry out relevant tests according to the following command sequence:
 sudo insmod ~/work/litex_m2sdr/software/kernel/litepcie.ko
 sudo chmod 666 /dev/m2sdr0

 Using dmesg, confirm that the following message appears:
 [2999.724330] litepcie: loading out-of-tree module taints kernel.
 [2999.725121] litepcie 0000:01:00.0: \x1b[1m[Probing device]\x1b[0m
 [2999.725143] litepcie 0000:01:00.0: enabling device (0000 -> 0002)
 [2999.745151] litepcie 0000:01:00.0: Version LiteX-M2SDR SoC / m2 variant / built on 2024-09-05 17:46:29
 [2999.745434] litepcie 0000:01:00.0: 1 MSI IRQs allocated.
 [2999.745500] litepcie 0000:01:00.0: Creating /dev/m2sdr0

 Execute ~/work/litex_m2sdr/software/user/m2sdr_util info, the following message will appear:
 [> FPGA/SoC Info:
 ------------------
 SoC Identifier : LiteX-M2SDR SoC / m2 variant / built on 2024-09-05 17:46:29.
 FPGA DNA: 0x0038cc0c7af9e854
 FPGA Temperature: 49.5 °C
 FPGA VCC-INT: 1.00 V
 FPGA VCC-AUX: 1.79 V
 FPGA VCC-BRAM: 1.00 V
 FPGA Status: Operational

 [> AD9361 Info:
 ---------------
 AD9361 Product ID: 000a
 AD9361 Temperature: 0.0 °C

 Execute ~/work/litex_m2sdr/software/user/m2sdr_rf, the following message will appear:
 Initializing SI5351 Clocking...
 Using internal XO for as SI5351 RefClk...
 Initializing AD9361 SPI...
 Initializing AD9361 RFIC...
 Setting Channel Mode to 2T2R.
 ad9361_init : AD936x Rev 2 successfully initialized
 Setting TX/RX Samplerate to 30.720000 MSPS.
 Setting TX/RX Bandwidth to 56.000000 MHz.
 Setting TX LO Freq to 98.200000 MHz.
 Setting RX LO Freq to 98.200000 MHz.
 Setting TX Gain to -20 dB.
 Setting RX Gain to 0 dB.
 Setting Loopback to 0
 Enabling 16-bit mode.

 At this point, the LiteX-M2SDR board installation has been completed.

F5OEO, 2025-03-31:
Which distro have you used ? armbian ? ubuntu community ?
On ubuntu, card is recognized, but no i/q from dma.

akioolin, 2025-03-31:
Hi, @F5OEO 

> Which distro have you used ? armbian ? ubuntu community ? On ubuntu, card is recognized, but no i/q from dma.

It seems not the distro probelm. In my case, the device tree setting is the key.

cd /boot/dtb/rockchip
sudo cp rk3588-orangepi-5-max.dtb rk3588-orangepi-5-max.orig.dtb
sudo dtc rk3588-orangepi-5-max.dtb -o rk3588-orangepi-5-max.dts
sudo joe rk3588-orangepi-5-max.dts
Find num-lanes in the pcie@fe150000 section, change its setting value from <0x04> to <0x02>, and save it.
sudo dtc rk3588-orangepi-5-max.dts -o rk3588-orangepi-5-max.dtb

After completion, reboot.

sync;sync;sync;sudo reboot

BR, Akio

F5OEO, 2025-03-31:
Without modifying dts, m2sdr is recognised.
Software and setting works properly, but I can't get I/Q through dma.
Have you also add iommu.passthrough=1 arm-smmu.disable=1 to cmdline ?

I also tried to change the dts as you mentionned, but nothing better. 
I am stuck for few days on that. Any advice is welcomed. 
I have orange pi reachable from outside through ssh if needed.

akioolin, 2025-04-04:
Hi, @F5OEO :

below is my test log. hope this can help you find out some useful data.

a. the num-lanes setting for pcie@fe150000 must be 0x02, for both litex_m2sdr_m2_pcie_x4.bin and litex_m2sdr_m2_pcie_x2.bin. 

the below is the kernel module loaded message:
[  580.761501] m2sdr: loading out-of-tree module taints kernel.
[  580.762335] m2sdr 0000:01:00.0: \x1b[1m[Probing device]\x1b[0m
[  580.762365] m2sdr 0000:01:00.0: enabling device (0000 -> 0002)
[  580.778686] m2sdr 0000:01:00.0: Version LiteX-M2SDR SoC / m2 variant / built on 2025-04-02 21:54:15
[  580.778951] m2sdr 0000:01:00.0: 1 MSI IRQs allocated.
[  580.779006] m2sdr 0000:01:00.0: Creating /dev/m2sdr0

b. change the num-lanes setting to 0x04 with litex_m2sdr_m2_pcie_x4.bin, the "RF controller: Xilinx Corporation Device 7024" could not be detected. 

c. with or without "iommu.passthrough=1 arm-smmu.disable=1" in kernel bootagrs seems no effect pcie detection. 

d. the OS which I'm using is 
Linux version 6.1.43-rockchip-rk3588 (root@orangepi5) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (G
NU Binutils for Ubuntu) 2.38) #1.0.0 SMP Mon Jul  8 11:54:40 CST 2024

e. as I know, the RK3588 based SBC are a big family, maybe the board which you are using and mine is not the same version, vendor. 

f. could you show me how to do the I/Q tests which you did? I just using m2sdr_record to record data. 

BR, Akio

F5OEO, 2025-04-10:
Thanks for your interesting remarks. I installed same distro.
I have change the dtb (0x4 -> 0x2)

I get 
```
dmesg |grep fe150000
[    6.087100] rk-pcie fe150000.pcie: invalid prsnt-gpios property in node
[    6.104767] rk-pcie fe150000.pcie: host bridge /pcie@fe150000 ranges:
[    6.104860] rk-pcie fe150000.pcie:       IO 0x00f0100000..0x00f01fffff -> 0x00f0100000
[    6.104887] rk-pcie fe150000.pcie:      MEM 0x00f0200000..0x00f0ffffff -> 0x00f0200000
[    6.104904] rk-pcie fe150000.pcie:      MEM 0x0900000000..0x093fffffff -> 0x0900000000
[    6.104958] rk-pcie fe150000.pcie: iATU unroll: enabled
[    6.104969] rk-pcie fe150000.pcie: iATU regions: 8 ob, 8 ib, align 64K, limit 8G
[    6.310844] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x0
[    6.336380] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x1
[    6.363045] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x1
[    6.579710] rk-pcie fe150000.pcie: PCIe Link up, LTSSM is 0x130011
[    6.579719] rk-pcie fe150000.pcie: PCIe Gen.2 x2 link up
[    6.579778] rk-pcie fe150000.pcie: PCI host bridge to bus 0000:00

 dmesg |grep "pci 0000:00"
[    6.579812] pci 0000:00:00.0: [1d87:3588] type 01 class 0x060400
[    6.579820] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x3fffffff]
[    6.579827] pci 0000:00:00.0: reg 0x14: [mem 0x00000000-0x3fffffff]
[    6.579832] pci 0000:00:00.0: reg 0x38: [mem 0x00000000-0x0000ffff pref]
[    6.579865] pci 0000:00:00.0: supports D1 D2
[    6.579869] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    6.584541] pci 0000:00:00.0: BAR 0: no space for [mem size 0x40000000]
[    6.584544] pci 0000:00:00.0: BAR 0: failed to assign [mem size 0x40000000]
[    6.584546] pci 0000:00:00.0: BAR 1: no space for [mem size 0x40000000]
[    6.584548] pci 0000:00:00.0: BAR 1: failed to assign [mem size 0x40000000]
[    6.584550] pci 0000:00:00.0: BAR 8: assigned [mem 0xf0200000-0xf02fffff]
[    6.584552] pci 0000:00:00.0: BAR 6: assigned [mem 0xf0300000-0xf030ffff pref]
[    6.584566] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
[    6.584568] pci 0000:00:00.0:   bridge window [mem 0xf0200000-0xf02fffff]

 lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd Device 3588 (rev 01)
0000:01:00.0 RF controller: Xilinx Corporation Device 7024
0003:30:00.0 PCI bridge: Rockchip Electronics Co., Ltd Device 3588 (rev 01)
0003:31:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)

dmesg |grep m2sdr
[   10.898810] m2sdr: loading out-of-tree module taints kernel.
[   10.900092] m2sdr 0000:01:00.0: \x1b[1m[Probing device]\x1b[0m
[   10.900136] m2sdr 0000:01:00.0: enabling device (0000 -> 0002)
[   10.920033] m2sdr 0000:01:00.0: Version LiteX-M2SDR SoC / m2 variant / built on 2024-08-21 14:28:31
[   10.920280] m2sdr 0000:01:00.0: 1 MSI IRQs allocated.
[   10.920401] m2sdr 0000:01:00.0: Creating /dev/m2sdr0

```
I flash [litex_m2sdr_m2_pcie_x2_2025_03_10.zip](https://github.com/user-attachments/files/19164857/litex_m2sdr_m2_pcie_x2_2025_03_10.zip) the with https://github.com/enjoy-digital/litex_m2sdr/issues/30 from 2025 03 10

But still no I/Q

 ```
./m2sdr_record 
SPEED(Gbps)    BUFFERS SIZE(MB)
      0.00          0         0
      0.00          0         0
      0.00          0         0
      0.00          0         0
      0.00          0         0
```

In order to test with Soapy you can use https://github.com/rxseger/rx_tools
And run rx_sdr -f 500e6 -s 10e6 - to see if i/q are displayed on stdout.

I could maybe try with x1 gateware, but at this time all seems the same as you instead m2 variant / built on 2024-08-21 14:28:31

If anyone could see what could be my issue , I am lost.
Regards
Evariste

akioolin, 2025-04-10:
Hi, @F5OEO :

Maybe you could run m2sdr_record as following way:

sudo ./m2sdr_record
or 

sudo chmod 666 /dev/m2sdr0
./m2sdr_record

BR, Akio

F5OEO, 2025-04-10:
Yes, my fault : 
```
SPEED(Gbps)    BUFFERS SIZE(MB)
      1.95       5993         0
      1.97      12025         0
      1.97      18057         0
      1.97      24089         0
      1.96      30113         0
      1.97      36145         0
      1.97      42177         0
```

But dmesg shows

```
243.671626] m2sdr 0000:01:00.0: Reading too late, 158 buffers lost
[  243.680185] m2sdr 0000:01:00.0: Reading too late, 191 buffers lost
[  243.688694] m2sdr 0000:01:00.0: Reading too late, 203 buffers lost
[  243.695752] m2sdr 0000:01:00.0: Reading too late, 1 buffers lost
[  288.979800] m2sdr 0000:01:00.0: Reading too late, 190 buffers lost
[  288.988340] m2sdr 0000:01:00.0: Reading too late, 209 buffers lost
[  288.997085] m2sdr 0000:01:00.0: Reading too late, 211 buffers lost
[  289.005492] m2sdr 0000:01:00.0: Reading too late, 208 buffers lost
[  289.014119] m2sdr 0000:01:00.0: Reading too late, 206 buffers lost
[  289.022602] m2sdr 0000:01:00.0: Reading too late, 207 buffers lost
[  289.031093] m2sdr 0000:01:00.0: Reading too late, 206 buffers lost
[  289.039584] m2sdr 0000:01:00.0: Reading too late, 206 buffers lost
[  289.048123] m2sdr 0000:01:00.0: Reading too late, 193 buffers lost
[  289.054553] m2sdr 0000:01:00.0: Reading too late, 40 buffers lost
[  315.262536] m2sdr 0000:01:00.0: Reading too late, 159 buffers lost
[  315.271065] m2sdr 0000:01:00.0: Reading too late, 187 buffers lost
[  315.278076] m2sdr 0000:01:00.0: Reading too late, 62 buffers lost
[  315.323967] m2sdr 0000:01:00.0: Reading too late, 157 buffers lost
[  315.329279] m2sdr 0000:01:00.0: Reading too late, 7 buffers lost
```

akioolin, 2025-04-10:
Hi, @F5OEO :

please check m2sdr_record options to minimize buffer lost rate.
and set the recorded file in the memory partition not in emmc / sdcard, etc. to reduce data transfer latency.

BR, Akio