enjoy-digital/litex_m2sdr #33: Failed to write to SI5351

improperaffo, 2024-10-21:
Dear all,
I have started playing with the LiteX (C variant). I have noticed that many commands that I execute result in the following output:
```
Initializing SI5351 Clocking...
Using internal XO for as SI5351 RefClk...
Failed to write to SI5351 at register 0x02
Failed to write to SI5351 at register 0x03
Failed to write to SI5351 at register 0x04
Failed to write to SI5351 at register 0x07
Failed to write to SI5351 at register 0x0F
Failed to write to SI5351 at register 0x10
Failed to write to SI5351 at register 0x11
Failed to write to SI5351 at register 0x12
Failed to write to SI5351 at register 0x13
Failed to write to SI5351 at register 0x14
Failed to write to SI5351 at register 0x15
Failed to write to SI5351 at register 0x16
Failed to write to SI5351 at register 0x17
Failed to write to SI5351 at register 0x22
Failed to write to SI5351 at register 0x23
Failed to write to SI5351 at register 0x24
Failed to write to SI5351 at register 0x25
Failed to write to SI5351 at register 0x26
Failed to write to SI5351 at register 0x27
Failed to write to SI5351 at register 0x28
Failed to write to SI5351 at register 0x29
Failed to write to SI5351 at register 0x2A
Failed to write to SI5351 at register 0x2B
Failed to write to SI5351 at register 0x2C
Failed to write to SI5351 at register 0x2D
Failed to write to SI5351 at register 0x2E
Failed to write to SI5351 at register 0x2F
Failed to write to SI5351 at register 0x30
Failed to write to SI5351 at register 0x31
Failed to write to SI5351 at register 0x32
Failed to write to SI5351 at register 0x33
Failed to write to SI5351 at register 0x34
Failed to write to SI5351 at register 0x35
Failed to write to SI5351 at register 0x36
Failed to write to SI5351 at register 0x37
Failed to write to SI5351 at register 0x38
Failed to write to SI5351 at register 0x39
Failed to write to SI5351 at register 0x3A
Failed to write to SI5351 at register 0x3B
Failed to write to SI5351 at register 0x3C
Failed to write to SI5351 at register 0x3D
Failed to write to SI5351 at register 0x3E
Failed to write to SI5351 at register 0x3F
Failed to write to SI5351 at register 0x40
Failed to write to SI5351 at register 0x41
Failed to write to SI5351 at register 0x42
Failed to write to SI5351 at register 0x43
Failed to write to SI5351 at register 0x44
Failed to write to SI5351 at register 0x45
Failed to write to SI5351 at register 0x46
Failed to write to SI5351 at register 0x47
Failed to write to SI5351 at register 0x48
Failed to write to SI5351 at register 0x49
Failed to write to SI5351 at register 0x4A
Failed to write to SI5351 at register 0x4B
Failed to write to SI5351 at register 0x4C
Failed to write to SI5351 at register 0x4D
Failed to write to SI5351 at register 0x4E
Failed to write to SI5351 at register 0x4F
Failed to write to SI5351 at register 0x50
Failed to write to SI5351 at register 0x51
Failed to write to SI5351 at register 0x52
Failed to write to SI5351 at register 0x53
Failed to write to SI5351 at register 0x54
Failed to write to SI5351 at register 0x55
Failed to write to SI5351 at register 0x56
Failed to write to SI5351 at register 0x57
Failed to write to SI5351 at register 0x58
Failed to write to SI5351 at register 0x59
Failed to write to SI5351 at register 0x5A
Failed to write to SI5351 at register 0x5B
Failed to write to SI5351 at register 0x95
Failed to write to SI5351 at register 0x96
Failed to write to SI5351 at register 0x97
Failed to write to SI5351 at register 0x98
Failed to write to SI5351 at register 0x99
Failed to write to SI5351 at register 0x9A
Failed to write to SI5351 at register 0x9B
Failed to write to SI5351 at register 0xA2
Failed to write to SI5351 at register 0xA3
Failed to write to SI5351 at register 0xA4
Failed to write to SI5351 at register 0xA5
Failed to write to SI5351 at register 0xA6
Failed to write to SI5351 at register 0xA7
Failed to write to SI5351 at register 0xA8
Failed to write to SI5351 at register 0xA9
Failed to write to SI5351 at register 0xAA
Failed to write to SI5351 at register 0xB7
Initializing AD9361 SPI...
Initializing AD9361 RFIC...
```
I am trying to observe some output from the SDR using a spectrum analyzer, but I do not get anything. Could this be a reason for it?

Thanks

enjoy-digital, 2024-10-21:
Hi @improperaffo,

the errors has also been reported by other user (intermittent error), we'll probably adjust the I2C parameter or replace the Core to fix this. I'm off for two weeks and travelling so will be able to look at this when back. As a quick test, it could be useful to increase the sleep here: https://github.com/enjoy-digital/litex_m2sdr/blob/main/software/user/libm2sdr/m2sdr_si5351_i2c.c#L23 or/and verify that `./m2sdr_util si5351_scan` is correctly detecting the chip.

enjoy-digital, 2024-11-20:
Hi @improperaffo, @dmd17,

it seems the issue was only happening when using > Gen2 X1 PCIe where I2C bitbanging was probably faster.  It would be interesting to test with the additional pullups on the I2C lines, added with https://github.com/enjoy-digital/litex_m2sdr/commit/c95bc4d3b027ae5d5135c07983e5d209d2cca295.

enjoy-digital, 2025-08-22:
With https://github.com/enjoy-digital/litex_m2sdr/pull/66, we are switching to LiteI2C to do the SI5351 I2C accesses which avoid bit banging and fixes the issues on a setup where the issues were often seen. I'll share new bitstreams in the next days and we'll be able to close this.

closed 2025-08-26 by enjoy-digital