enjoy-digital/litex_m2sdr #34: Unable to send via Soapy

dmd17, 2024-11-04:
added some debug statements in LiteXM2SDRStreaming.cpp

at the start of writeStream
2024-11-04T00:24:23.093353 [RF      ] [D] Soapy: writeStream: numElems 11520

in acquireWriteBuffer after the first call to the DMA engine
2024-11-04T00:24:23.093361 [RF      ] [D] Soapy: acquireWriteBuffer: _tx_stream.hw_count 21856, _tx_stream.sw_count 0, _dma_mmap_info.dma_tx_buf_count 256

in acquireWriteBuffer after the second call to the DMA engine
2024-11-04T00:24:23.093364 [RF      ] [D] Soapy: acquireWriteBuffer: _tx_stream.hw_count 21856, _tx_stream.sw_count 0, _dma_mmap_info.dma_tx_buf_count 256

in writeStream after the call to acquireWriteBuffer
2024-11-04T00:24:23.093368 [RF      ] [D] Soapy: writeStream: could not acquire write buffer -7

enjoy-digital, 2024-11-04:
Thanks @dmd17, this will be investigated in the next days.

enjoy-digital, 2024-11-04:
@dmd17: Would you mind sharing your additional debug traces and high level test to easily reproduce?

dmd17, 2024-11-04:
I was able to send using SoapySDRUtil

SoapySDRUtil --args="driver=LiteXM2SDR" --rate=30720000  --direction=TX

however, it is unstable and at some point it crashes when i try to use 2 channels.

dmd17, 2024-11-04:
#####################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] SoapyLiteXM2SDR initializing...
[INFO] Received arguments:
[INFO]   bitmode: 16
[INFO]   device: LiteX-M2SDR
[INFO]   driver: LiteXM2SDR
[INFO]   identification: LiteX-M2SDR SoC / m2 variant / built on 2024-09-05 17:46:29
[INFO]   label: LiteX-M2SDR /dev/m2sdr0 c45067470485c LiteX-M2SDR SoC / m2 variant / built on 2024-09-05 17:46:29
[INFO]   oversampling: 0
[INFO]   path: /dev/m2sdr0
[INFO]   serial: c45067470485c
[INFO]   version: 1234
[INFO] Opened devnode /dev/m2sdr0, serial c45067470485c
Failed to write to SI5351 at register 0x02
Failed to write to SI5351 at register 0x03
Failed to write to SI5351 at register 0x04
Failed to write to SI5351 at register 0x07
Failed to write to SI5351 at register 0x0F
Failed to write to SI5351 at register 0x10
Failed to write to SI5351 at register 0x11
Failed to write to SI5351 at register 0x12
Failed to write to SI5351 at register 0x13
Failed to write to SI5351 at register 0x14
Failed to write to SI5351 at register 0x15
Failed to write to SI5351 at register 0x16
Failed to write to SI5351 at register 0x17
Failed to write to SI5351 at register 0x22
Failed to write to SI5351 at register 0x23
Failed to write to SI5351 at register 0x24
Failed to write to SI5351 at register 0x25
Failed to write to SI5351 at register 0x26
Failed to write to SI5351 at register 0x27
Failed to write to SI5351 at register 0x28
Failed to write to SI5351 at register 0x29
Failed to write to SI5351 at register 0x2A
Failed to write to SI5351 at register 0x2B
Failed to write to SI5351 at register 0x2C
Failed to write to SI5351 at register 0x2D
Failed to write to SI5351 at register 0x2E
Failed to write to SI5351 at register 0x2F
Failed to write to SI5351 at register 0x30
Failed to write to SI5351 at register 0x31
Failed to write to SI5351 at register 0x32
Failed to write to SI5351 at register 0x33
Failed to write to SI5351 at register 0x34
Failed to write to SI5351 at register 0x35
Failed to write to SI5351 at register 0x36
Failed to write to SI5351 at register 0x37
Failed to write to SI5351 at register 0x38
Failed to write to SI5351 at register 0x39
Failed to write to SI5351 at register 0x3A
Failed to write to SI5351 at register 0x3B
Failed to write to SI5351 at register 0x3C
Failed to write to SI5351 at register 0x3D
Failed to write to SI5351 at register 0x3E
Failed to write to SI5351 at register 0x3F
Failed to write to SI5351 at register 0x40
Failed to write to SI5351 at register 0x41
Failed to write to SI5351 at register 0x42
Failed to write to SI5351 at register 0x43
Failed to write to SI5351 at register 0x44
Failed to write to SI5351 at register 0x45
Failed to write to SI5351 at register 0x46
Failed to write to SI5351 at register 0x47
Failed to write to SI5351 at register 0x48
Failed to write to SI5351 at register 0x49
Failed to write to SI5351 at register 0x4A
Failed to write to SI5351 at register 0x4B
Failed to write to SI5351 at register 0x4C
Failed to write to SI5351 at register 0x4D
Failed to write to SI5351 at register 0x4E
Failed to write to SI5351 at register 0x4F
Failed to write to SI5351 at register 0x50
Failed to write to SI5351 at register 0x51
Failed to write to SI5351 at register 0x52
Failed to write to SI5351 at register 0x53
Failed to write to SI5351 at register 0x54
Failed to write to SI5351 at register 0x55
Failed to write to SI5351 at register 0x56
Failed to write to SI5351 at register 0x57
Failed to write to SI5351 at register 0x58
Failed to write to SI5351 at register 0x59
Failed to write to SI5351 at register 0x5A
Failed to write to SI5351 at register 0x5B
Failed to write to SI5351 at register 0x95
Failed to write to SI5351 at register 0x96
Failed to write to SI5351 at register 0x97
Failed to write to SI5351 at register 0x98
Failed to write to SI5351 at register 0x99
Failed to write to SI5351 at register 0x9A
Failed to write to SI5351 at register 0x9B
Failed to write to SI5351 at register 0xA2
Failed to write to SI5351 at register 0xA3
Failed to write to SI5351 at register 0xA4
Failed to write to SI5351 at register 0xA5
Failed to write to SI5351 at register 0xA6
Failed to write to SI5351 at register 0xA7
Failed to write to SI5351 at register 0xA8
Failed to write to SI5351 at register 0xA9
Failed to write to SI5351 at register 0xAA
Failed to write to SI5351 at register 0xB7
ad9361_init : AD936x Rev 2 successfully initialized
[INFO] SoapyLiteXM2SDR initialization complete
Stream format: CF32
Num channels: 1
Element size: 8 bytes
Begin TX rate test at 30.72 Msps
Starting stream loop, press Ctrl+C to exit...

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7f014d2 in SoapyLiteXM2SDR::interleaveCF32(void const*, void*, unsigned int, unsigned long) ()
   from /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.8/libSoapyLiteXM2SDR.so
(gdb) bt
#0  0x00007ffff7f014d2 in SoapyLiteXM2SDR::interleaveCF32(void const*, void*, unsigned int, unsigned long) ()
   from /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.8/libSoapyLiteXM2SDR.so
#1  0x00007ffff7f03143 in SoapyLiteXM2SDR::writeStream(SoapySDR::Stream*, void const* const*, unsigned long, int&, long long, long) () from /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.8/libSoapyLiteXM2SDR.so
#2  0x0000555555564e88 in runRateTestStreamLoop (device=0x555555580730, stream=<optimized out>, 
    direction=<optimized out>, numChans=<optimized out>, elemSize=<optimized out>)
    at /usr/src/soapysdr-0.8.1-4build1/apps/SoapyRateTest.cpp:58
#3  0x000055555556580f in SoapySDRRateTest (argStr="driver=LiteXM2SDR", sampleRate=sampleRate@entry=30720000, 
    channelStr="2", directionStr="TX") at /usr/src/soapysdr-0.8.1-4build1/apps/SoapyRateTest.cpp:160
#4  0x00005555555592bb in main (argc=<optimized out>, argv=<optimized out>)
    at /usr/src/soapysdr-0.8.1-4build1/apps/SoapySDRUtil.cpp:387

enjoy-digital, 2024-11-05:
Thanks for the additional info @dmd17, would you mind also sharing some info about the Host machine?

dmd17, 2024-11-05:
I hacked up SoapySDRUtil and created a test of transmit and receive at the same time
https://github.com/pvnis/SoapyTest

apparently putting a sleep between the activation of the RX and TX streams fixed the problem. 

please try to reproduce, with and without the sleep.

./SoapySDRUtil --args="driver=LiteXM2SDR" --frequency=3489420000 --rate=30720000 --channels=0,1

dmd17, 2024-11-05:
however, when i let it run longer, it gets into a bad state again.

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] SoapyLiteXM2SDR initializing...
[INFO] Received arguments:
[INFO]   bitmode: 16
[INFO]   device: LiteX-M2SDR
[INFO]   driver: LiteXM2SDR
[INFO]   identification: LiteX-M2SDR SoC / m2 variant / built on 2024-09-05 17:46:29
[INFO]   label: LiteX-M2SDR /dev/m2sdr0 c45067470485c LiteX-M2SDR SoC / m2 variant / built on 2024-09-05 17:46:29
[INFO]   oversampling: 0
[INFO]   path: /dev/m2sdr0
[INFO]   serial: c45067470485c
[INFO]   version: 1234
[INFO] Opened devnode /dev/m2sdr0, serial c45067470485c
Failed to write to SI5351 at register 0x02
Failed to write to SI5351 at register 0x03
Failed to write to SI5351 at register 0x04
Failed to write to SI5351 at register 0x07
Failed to write to SI5351 at register 0x0F
Failed to write to SI5351 at register 0x10
Failed to write to SI5351 at register 0x11
Failed to write to SI5351 at register 0x12
Failed to write to SI5351 at register 0x13
Failed to write to SI5351 at register 0x14
Failed to write to SI5351 at register 0x15
Failed to write to SI5351 at register 0x16
Failed to write to SI5351 at register 0x17
Failed to write to SI5351 at register 0x22
Failed to write to SI5351 at register 0x23
Failed to write to SI5351 at register 0x24
Failed to write to SI5351 at register 0x25
Failed to write to SI5351 at register 0x26
Failed to write to SI5351 at register 0x27
Failed to write to SI5351 at register 0x28
Failed to write to SI5351 at register 0x29
Failed to write to SI5351 at register 0x2A
Failed to write to SI5351 at register 0x2B
Failed to write to SI5351 at register 0x2C
Failed to write to SI5351 at register 0x2D
Failed to write to SI5351 at register 0x2E
Failed to write to SI5351 at register 0x2F
Failed to write to SI5351 at register 0x30
Failed to write to SI5351 at register 0x31
Failed to write to SI5351 at register 0x32
Failed to write to SI5351 at register 0x33
Failed to write to SI5351 at register 0x34
Failed to write to SI5351 at register 0x35
Failed to write to SI5351 at register 0x36
Failed to write to SI5351 at register 0x37
Failed to write to SI5351 at register 0x38
Failed to write to SI5351 at register 0x39
Failed to write to SI5351 at register 0x3A
Failed to write to SI5351 at register 0x3B
Failed to write to SI5351 at register 0x3C
Failed to write to SI5351 at register 0x3D
Failed to write to SI5351 at register 0x3E
Failed to write to SI5351 at register 0x3F
Failed to write to SI5351 at register 0x40
Failed to write to SI5351 at register 0x41
Failed to write to SI5351 at register 0x42
Failed to write to SI5351 at register 0x43
Failed to write to SI5351 at register 0x44
Failed to write to SI5351 at register 0x45
Failed to write to SI5351 at register 0x46
Failed to write to SI5351 at register 0x47
Failed to write to SI5351 at register 0x48
[... 778 lines cut ...]
40.2156 Msps	643.45 MBps - Dir 0	Underflows 63
40.2087 Msps	643.339 MBps - Dir 1	Overflows 24
40.2156 Msps	643.45 MBps - Dir 0	Underflows 95
40.2087 Msps	643.34 MBps - Dir 1	Overflows 24
40.2156 Msps	643.45 MBps - Dir 0	Underflows 95
40.2087 Msps	643.34 MBps - Dir 1	Overflows 24
40.2157 Msps	643.451 MBps - Dir 0	Underflows 95
40.2088 Msps	643.341 MBps - Dir 1	Overflows 24
40.2157 Msps	643.451 MBps - Dir 0	Underflows 95
40.2089 Msps	643.342 MBps - Dir 1	Overflows 24
40.2157 Msps	643.451 MBps - Dir 0	Underflows 95
40.2089 Msps	643.342 MBps - Dir 1	Overflows 24
40.2158 Msps	643.452 MBps - Dir 0	Underflows 95
40.209 Msps	643.343 MBps - Dir 1	Overflows 24
40.2158 Msps	643.453 MBps - Dir 0	Underflows 95
40.209 Msps	643.344 MBps - Dir 1	Overflows 24
40.2158 Msps	643.453 MBps - Dir 0	Underflows 95
40.209 Msps	643.345 MBps - Dir 1	Overflows 24
40.2158 Msps	643.453 MBps - Dir 0	Underflows 127
40.2089 Msps	643.343 MBps - Dir 1	Overflows 25
40.2158 Msps	643.452 MBps - Dir 0	Underflows 127
40.209 Msps	643.343 MBps - Dir 1	Overflows 25
40.2158 Msps	643.452 MBps - Dir 0	Underflows 127
40.209 Msps	643.344 MBps - Dir 1	Overflows 25
40.2157 Msps	643.452 MBps - Dir 0	Underflows 127

enjoy-digital, 2024-11-05:
@dmd17: Thanks, can you share the host machine used?

dmd17, 2024-11-05:
https://arkpwa-stg.intel.com/content/www/us/en/ark/products/121637/intel-nuc-12-pro-kit-nuc12wski7.html

dmd17, 2024-11-05:
at a lower rate it seems more stable. no underflows at 10MHz sampling rate.

dmd17, 2024-11-07:
Any update on this? Did you get similar behavior?

enjoy-digital, 2024-11-07:
Sorry I haven't been able to look at it yet, will try soon and provide update.

trabucayre, 2024-11-12:
Hi. I have started to fix this issue.
My board has a gateware build around  2024-08-21 (so a bit older than you). No issues with si5351.

I try first to find a fix for the crash when the sleep is commented: it seems introduced by https://github.com/enjoy-digital/litex_m2sdr/commit/1a8df3a421362c3be0b94680a0609583cef74227
before this commit, no needs to add the sleep and no underflows

For SI5351 I will try to find in a second time.

trabucayre, 2024-11-12:
This commit  https://github.com/enjoy-digital/litex_m2sdr/commit/3a40f06a579c9f7a8d8845d61d286aee5dc070be fix crash.

For underflow I am able to observe this too, but by changing cpu scaling governor I have no more underflows:
```bash
cpu_path=/sys/devices/system/cpu
pstate=$cpu_path/intel_pstate/status

echo "active" > $pstate

for i in `ls -d $cpu_path/cpu[0-9]*`; do
    governor_file="$i/cpufreq/scaling_governor"     
    echo "performance" > $governor_file     
done
```

trabucayre, 2024-11-12:
For si5351: I have tried with the bitstream you use and with a bitstream build using latest commit and I'm unable to reproduces your issue (and in fact with both bitstreams I don't needs to switch cpu in performance mode to avoid having underflow).

trabucayre, 2024-11-18:
@dmd17 any news? Is this issue fully/partially fixed? Thanks!

dmd17, 2024-11-19:
traveling for work... i'll check it end of this week or early next week.

enjoy-digital, 2025-02-17:
Closing since no feedback and the issue should be fixed. Please reopen @dmd17 in case you still have issues.

closed 2025-02-17 by enjoy-digital