enjoy-digital/litex_m2sdr #35: SoapySDR::Device::getHardwareTime(const std::string &)

dmd17, 2024-11-08:
How can we implement this function? We need to read it from the SI5351?

enjoy-digital, 2024-11-08:
@dmd17:  This could probably be done with something like this:

```diff
diff --git a/litex_m2sdr.py b/litex_m2sdr.py
index 7f19f84..9c9698c 100755
--- a/litex_m2sdr.py
+++ b/litex_m2sdr.py
@@ -149,6 +149,8 @@ class BaseSoC(SoCMini):
         SoCMini.__init__(self, platform, sys_clk_freq,
             ident         = f"LiteX-M2SDR SoC / {variant} variant / built on",
             ident_version = True,
+            with_timer    = True,
+            timer_uptime  = True,
         )

```

Which will give you new registers:
```diff
+#define CSR_TIMER0_UPTIME_LATCH_ADDR 0x4820L
+#define CSR_TIMER0_UPTIME_LATCH_SIZE 1
+#define CSR_TIMER0_UPTIME_CYCLES_ADDR 0x4824L
+#define CSR_TIMER0_UPTIME_CYCLES_SIZE 2
```

That you could use to get uptime of the hardware (in` sys_clk` cycles): Write 1 to LATCH then read CYCLES.

dmd17, 2024-11-08:
Thanks. I'm building a new bitstream. I hope I don't brick it. :)

dmd17, 2024-11-08:
#define CSR_TIMER0_UPTIME_LATCH_ADDR 0x6820L
#define CSR_TIMER0_UPTIME_LATCH_SIZE 1
#define CSR_TIMER0_UPTIME_CYCLES_ADDR 0x6824L
#define CSR_TIMER0_UPTIME_CYCLES_SIZE 2

the addresses are different than your example - is that ok?

enjoy-digital, 2024-11-08:
@dmd17: If you use the generated `csr.h`, it will not be an issue. If this feature is integrated, we'll use a fixed CSR mapping as for the other peripherals.

dmd17, 2024-11-09:
It works. Thank you.

enjoy-digital, 2024-11-09:
@dmd17: Great, would you mind contributing it as a PR? I'll then make sure to use a fixed CSR for the Timer.

enjoy-digital, 2025-02-17:
This has been re-implemented differently with https://github.com/enjoy-digital/litex_m2sdr/pull/44 and also with setHardwareTime. The clock now use for getHardwareTime is now generated from the same clock source used for the RF (SI5351) and not from an onboard oscillator, so will have the same drift than  the RF.

closed 2025-02-17 by enjoy-digital