enjoy-digital/litex_m2sdr #37: Raspberry Pi XDMA Driver Seg Fault

AEW2015, 2024-11-20:
Installing driver caused a seg fault (likely due to DMA TLB packets), but driver still installed.
I have seen issues with custom TLB packets and the Broadcom PCIe root complex before.
Will explore!

ssh
```
/litex_m2sdr/software/kernel$ sudo ARCH=arm64 ./init.sh
[sudo] password for pi:
Module not compiled

Message from syslogd@ubuntu at Nov 20 14:54:24 ...
 kernel:[  137.230163] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP
Segmentation fault

Message from syslogd@ubuntu at Nov 20 14:54:24 ...
 kernel:[  137.508224] Code: 0b000280 97eba1c6 f9403801 b4ffff01 (d4210000)

$ lspci -vvv
01:00.0 RF controller: Xilinx Corporation Device 7024
        Subsystem: Xilinx Corporation Device 0007
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Region 0: Memory at 600000000 (32-bit, non-prefetchable) [size=1M]
        Capabilities: <access denied>
        Kernel driver in use: litepcie
```

Serial Ouput

```
[  137.213507] litepcie 0000:01:00.0: Failed to allocate dma buffe         s
[  137.219745] litepcie 0000:01:00.0: Failed to allocate DMA
[  137.225477] kernel BUG at drivers/pci/msi.c:396!
[  137.230163] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP
[  137.235725] Modules linked in: litepcie(OE+) dm_multipath scsi_dh_rdac scsi_         h_emc scsi_dh_alua binfmt_misc bcm2835_codec(CE) bcm2835_isp(CE) bcm2835_v4l2(C         ) v4l2_mem2mem bcm2835_mmal_vchiq(CE) videobuf2_vmalloc videobuf2_dma_contig ra         pberrypi_hwmon videobuf2_memops videobuf2_v4l2 videobuf2_common snd_bcm2835(CE)         snd_pcm videodev snd_timer mc vc_sm_cma(CE) snd rpivid_mem uio_pdrv_genirq uio          ch_fq_codel drm ramoops reed_solomon ip_tables x_tables autofs4 btrfs zstd_comp         ess raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx x         r xor_neon raid6_pq libcrc32c raid1 raid0 multipath linear dwc2 udc_core spidev         crct10dif_ce phy_generic aes_neon_bs aes_neon_blk crypto_simd cryptd
[  137.298391] CPU: 2 PID: 1798 Comm: insmod Tainted: G         C OE     5.4.0-         101-raspi #113-Ubuntu
[  137.307480] Hardware name: Raspberry Pi Compute Module 4 Rev 1.1 (DT)
[  137.314011] pstate: a0400005 (NzCv daif +PAN -UAO)
[  137.318873] pc : free_msi_irqs+0x174/0x198
[  137.323022] lr : free_msi_irqs+0x16c/0x198
[  137.327169] sp : ffff800010fc36d0
[  137.330524] x29: ffff800010fc36d0 x28: 00000000000020f8
[  137.335908] x27: 0000000000000000 x26: ffff2a3a384e9e00
[  137.341291] x25: ffff2a3a38b680c0 x24: 00000000fffffff4
[  137.346674] x23: ffff2a3a38b688c0 x22: ffff2a3a3bcf3000
[  137.352058] x21: ffff2a3a3bcf32c8 x20: 0000000000000000
[  137.357441] x19: ffff2a3a3b087c80 x18: 0000000000000000
[  137.362823] x17: 0000000000000000 x16: ffffcbec10687ac0
[  137.368207] x15: 0000000000000004 x14: 74696c3d4d455453
[  137.373591] x13: 0000000000000000 x12: ffff2a3a36acb8d8
[  137.378974] x11: ffff2a3a36acb6c0 x10: 0000000000000040
[  137.384357] x9 : ffffcbec1182ec10 x8 : ffffcbec1182ec08
[  137.389740] x7 : ffff2a3a3cc01180 x6 : 0000000000000000
[  137.395124] x5 : ffff2a3a3cc00ff8 x4 : 0000000000000000
[  137.400507] x3 : 0000000000000000 x2 : 0000000000000000
[  137.405890] x1 : ffff2a3a3b087380 x0 : ffff2a3a3a6f0400
[  137.411274] Call trace:
[  137.413749]  free_msi_irqs+0x174/0x198
[  137.417546]  pci_disable_msi+0x110/0x138
[  137.421519]  pci_free_irq_vectors+0x2c/0x38
[  137.425765]  litepcie_pci_probe+0x5b4/0x5fc [litepcie]
[  137.430973]  pci_device_probe+0xe8/0x198
[  137.434946]  really_probe+0xec/0x448
[  137.438566]  driver_probe_device+0xe8/0x138
[  137.442803]  device_driver_attach+0x78/0x80
[  137.447041]  __driver_attach+0xcc/0x188
[  137.450926]  bus_for_each_dev+0x84/0xe0
[  137.454810]  driver_attach+0x34/0x40
[  137.458430]  bus_add_driver+0x14c/0x230
[  137.462314]  driver_register+0x68/0x118
[  137.466198]  __pci_register_driver+0x5c/0x70
[  137.470525]  litepcie_module_init+0xbc/0x1000 [litepcie]
[  137.475908]  do_one_initcall+0x54/0x2e0
[  137.479793]  do_init_module+0x50/0x210
[  137.483589]  load_module+0x103c/0x1358
[  137.487385]  __do_sys_finit_module+0xac/0x118
[  137.491799]  __arm64_sys_finit_module+0x2c/0x38
[  137.496389]  el0_svc_common.constprop.0+0x84/0x230
[  137.501244]  el0_svc_handler+0x38/0xa0
[  137.505040]  el0_svc+0x10/0x140
[  137.508224] Code: 0b000280 97eba1c6 f9403801 b4ffff01 (d4210000)
[  137.514403] ---[ end trace ad18507a6421dc02 ]---
[  137.519081] note: insmod[1798] exited with preempt_count 1
[  137.525133] ------------[ cut here ]------------
[  137.529837] WARNING: CPU: 2 PID: 0 at kernel/rcu/tree.c:569 rcu_idle_enter+0         84/0x90
[  137.537691] Modules linked in: litepcie(OE+) dm_multipath scsi_dh_rdac scsi_         h_emc scsi_dh_alua binfmt_misc bcm2835_codec(CE) bcm2835_isp(CE) bcm2835_v4l2(C         ) v4l2_mem2mem bcm2835_mmal_vchiq(CE) videobuf2_vmalloc videobuf2_dma_contig ra         pberrypi_hwmon videobuf2_memops videobuf2_v4l2 videobuf2_common snd_bcm2835(CE)         snd_pcm videodev snd_timer mc vc_sm_cma(CE) snd rpivid_mem uio_pdrv_genirq uio          ch_fq_codel drm ramoops reed_solomon ip_tables x_tables autofs4 btrfs zstd_comp         ess raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx x         r xor_neon raid6_pq libcrc32c raid1 raid0 multipath linear dwc2 udc_core spidev         crct10dif_ce phy_generic aes_neon_bs aes_neon_blk crypto_simd cryptd
[  137.600400] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G      D  C OE     5.4.0-         101-raspi #113-Ubuntu
[  137.609500] Hardware name: Raspberry Pi Compute Module 4 Rev 1.1 (DT)
[  137.616041] pstate: 204003c5 (nzCv DAIF +PAN -UAO)
[  137.620909] pc : rcu_idle_enter+0x84/0x90
[  137.624983] lr : do_idle+0x210/0x2a0
[  137.628609] sp : ffff80001009bf20
[  137.631972] x29: ffff80001009bf20 x28: 0000000046a1c2e4
[  137.637366] x27: 0000000000000000 x26: 0000000000000000
[  137.642760] x25: 0000000000000000 x24: ffffcbec11809a2c
[  137.648154] x23: ffff2a3a3d22da00 x22: ffffcbec11524178
[  137.653549] x21: ffffcbec118099f8 x20: 0000000000000002
[  137.658943] x19: ffffcbec11809978 x18: 0000000000000010
[  137.664337] x17: 0000000000000000 x16: ffffcbec10687ac0
[  137.669731] x15: ffff2a3a384ea338 x14: 0720072007200720
[  137.675126] x13: 000000000007b1c8 x12: 0000000000000001
[  137.680520] x11: ffffcbec110bf7b8 x10: 00000000000019d0
[  137.685914] x9 : ffff80001009be60 x8 : ffff2a3a3d22f430
[  137.691308] x7 : 0000000000000000 x6 : 0000000000000001
[  137.696701] x5 : 0000000000000001 x4 : 0000000000006814
[  137.702095] x3 : 4000000000000002 x2 : 4000000000000000
[  137.707489] x1 : ffffcbec11526400 x0 : ffff2a3a3fbbd400
[  137.712883] Call trace:
[  137.715368]  rcu_idle_enter+0x84/0x90
[  137.719088]  do_idle+0x210/0x2a0
[  137.722366]  cpu_startup_entry+0x2c/0x78
[  137.726353]  secondary_start_kernel+0x1a4/0x1d0
[  137.730951] ---[ end trace ad18507a6421dc03 ]---
```

enjoy-digital, 2024-11-20:
Hi @AEW2015,

are you planning to use the board with LitePCIe or XDMA? This has only been experimented with LitePCIe for now in https://github.com/enjoy-digital/litex_m2sdr/issues/18.

AEW2015, 2024-11-20:
Sorry, that was a mistype.  It was the LitePCIe driver from the OrangePi example

smunaut, 2025-04-29:
Check the page size of your kernel. Depending on which distribution you might have 64kb page size.

trabucayre, 2025-04-29:
@smunaut True.
See https://github.com/enjoy-digital/litex_m2sdr/blob/main/doc/hosts/raspberry-pi-5.md#tweaking-the-system for details on howto configure your board.

enjoy-digital, 2026-10-21:
Closing since the driver issue has been fixed and there was no further feedback. Feel free to reopen if the issue is still seen.

closed 2026-10-21 by enjoy-digital