enjoy-digital/litex_m2sdr #43: Are the m2sdr schematics available?

neerajkc, 2024-12-28:
Hi, could you please release the schematics for this board. Want to know what parts, pin mappings, speedgrade, DDR chip and capacity is used. Also, please release supporting files like vivado project, constraints file, etc for any custom HDL development. Thanks!

enjoy-digital, 2024-12-28:
Hi @neerajkc,

for now the schematic are not available (common decision with my hardware partner).  By looking at the shared file, you'll see that almost the information you are requesting are present: ex parts, pin mappings, speedgrade. This project is built with LiteX framework, that automatically generates the Vivado project and constraints file, since these files are generated during the build, they are not commited/present in the repository.

neerajkc, 2024-12-30:
@enjoy-digital does the board have onboard DDR RAM, if yes then how much? Is it possible to use XDMA with this board? I intend to use this board with Nvidia Jetson, do you see any issue using litex on a Jetson board?

enjoy-digital, 2024-12-30:
There’s no DDR RAM on the board because LitePCIe doesn’t require on-FPGA buffering; all buffering is handled by the host’s RAM. It’s a standard Artix-7 FPGA, so XDMA would be possible, but we’ve found LitePCIe to be a more advantageous replacement. Over the past few years, we’ve assisted multiple clients in switching from XDMA to LitePCIe due to various issues with XDMA. Consequently, we only provide support for our LiteX/LitePCIe design. Testing with Jetson devices is on our roadmap for early 2025, although other users have already confirmed successful operation.