enjoy-digital/litex_m2sdr #49: Vivado 2024.2 Complition errors/critical warnings for PCI-e lanes =2.

Godtec, 2025-03-18:
Hello All:

After getting my LiteXM2SDR board to work :) YEAH
I am looking to work on custom gateware code for the FPGA. After several attempts, I get Critical warnings with the build process of the new gateware. I am using the latest version Vivado 2024.2 64 bit.
I am not very familiar with Vivado, so please bear with me.. I also verified that I have license for the PCIe IP block within Vivado for the XC7A200Tfbg-484-3 device.

Here are some of the noteable errors/critical warnings:

`CRITICAL WARNING: [Vivado 12-5447] synth_ip is not supported in project mode, please use non-project mode.`

I read later that the Command generated that was ran with the "Mode out_of_context" used. I am assuming we can safely ignore.

Plus these repeated Clock generation warnings. (I do NOT think we can ignore these warnings)

> WARNING: [Vivado 12-627] No clocks matched 'si5351_clk1'. [/home/mikek/Documents/Cyclone5_SOC/Litex_Release_March_2025/litex_m2sdr/build/litex_m2sdr_m2_pcie/gateware/litex_m2sdr_m2_pcie.xdc:389]

> INFO: [Vivado 12-626] No clocks found. Please use 'create_clock' or 'create_generated_clock' command to create clocks. [/home/mikek/Documents/Cyclone5_SOC/Litex_Release_March_2025/litex_m2sdr/build/litex_m2sdr_m2_pcie/gateware/litex_m2sdr_m2_pcie.xdc:389]

> CRITICAL WARNING: [Vivado 12-4739] set_clock_groups:No valid object(s) found for '-group [get_clocks si5351_clk1]'. [/home/mikek/Documents/Cyclone5_SOC/Litex_Release_March_2025/litex_m2sdr/build/litex_m2sdr_m2_pcie/gateware/litex_m2sdr_m2_pcie.xdc:389]

> Resolution: Check if the specified object(s) exists in the current design. If it does, ensure that the correct design hierarchy was specified for the object. If you are working with clocks, make sure create_clock was used to create the clock object before it is referenced.

Repeated ~14 times for other clocks.  I believe that the command create_clock was not created for port clk1, I have adjusted the xdc, by adding the clock info for si5351_clk1, by replicating clk0, and it has seemed to solved that issue. but the next error is still there:

> CRITICAL WARNING: [Vivado 12-4739] set_clock_groups:No valid object(s) found for '-group [get_clocks -include_generated_clocks -of [get_nets pcie_clk]]'. [/home/mikek/Documents/Cyclone5_SOC/Litex_Release_March_2025/litex_m2sdr/build/litex_m2sdr_m2_pcie/gateware/litex_m2sdr_m2_pcie.xdc:371]
> Resolution: Check if the specified object(s) exists in the current design. If it does, ensure that the correct design hierarchy was specified for the object. If you are working with clocks, make sure create_clock was used to create the clock object before it is referenced.

Also, Further along in the build I got this Critical Warning:

> CRITICAL WARNING: [Vivado 12-1411] Cannot set LOC property of ports, Could not legally place instance
> pcie_x1_m2_rx_p_IBUF_inst
>  at C9 (IPAD_X1Y60) since it belongs to a shape containing instance
>  pcie_s7/inst/inst/gt_top_i/pipe_wrapper_i/pipe_lane[0].gt_wrapper_i/gtp_channel.gtpe2_channel_i. The shape requires relative placement between
>  pcie_x1_m2_rx_p_IBUF_inst and
> pcie_s7/inst/inst/gt_top_i/pipe_wrapper_i/pipe_lane[0].gt_wrapper_i/gtp_channel.gtpe2_channel_i
>  that can not be honoured because it would result in an invalid location for
> pcie_s7/inst/inst/gt_top_i/pipe_wrapper_i/pipe_lane[0].gt_wrapper_i/gtp_channel.gtpe2_channel_i. [/home/mikek/Documents/Cyclone5_SOC/Litex_Release_March_2025/litex_m2sdr/build/litex_m2sdr_m2_pcie/gateware/litex_m2sdr_m2_pcie.xdc:74]

Again, I am not sure if I can safely ignore this? Or is this was note because of the clocks not being created.

Any help would be appreciated!!
Thanks Again!
Mike Kennedy
VA3TEC

[vivado.log](https://github.com/user-attachments/files/19322482/vivado.log)

enjoy-digital, 2025-03-19:
Hi @Godtec,

thanks for the feedback. These warning are probably related to recent changes/added features and I'll redo a pass on the warning and remove all un-wanted warnings.

colin-rushton-qss, 2025-08-07:
I'm also having the same/similar issue:

Vivado blows up on the parsing of `gateware/litex_m2sdr_m2_pcie_x1.xdc` due to `cannot set LOC property of ports, the positive port side of a differential pair cannot be placed on a negative package pin`, `set_clock_groups: no valid objects found for '--group...`, and the `hs_err_pid#####.log` has a stacktrace that has no additional information other than that it was using vivado XDC libs and an unexpected error has occurred.

Not sure if they are related but during the initial steps calling Vivado it warns `Attempt to change 'Component_Name' from 'pcie_s7' to 'pcie' is not allowed and is ignored` followed by some warnings during synthesis about `system function call 'time' not supported`.  There is then during netlist optimization a warning about `Creating clock pcie_s7_dclk with 2 sources`, then a few warnings about unconnected ports referencing a half dozen different modules a lot of them clock or ready flags it appears all of these in relation to `pcie_s7`.  Then the actual fatal errors which are almost entirely warnings for various clocks and not finding matching nets or on the no valid objects for clock groups errors.  Gets past synthesis there claiming 0 errors, 0 criticals, and a bunch of warnings despite there being a large number of criticals, and immediately hits errors in parsing the XDC file with the LOC properties issue which then finally fails the whole process.

This is with running `./litex_m2sdr.py --with-pcie --variant=m2 --build --load`

Frankly, I don't know enough about litex or vivado to diagnose and fix this myself, as I was primarily hoping to stay on the RF side of the project and this seemed like a good candidate but being unable to even build the current main branch is a problem.

enjoy-digital, 2025-08-07:
Hi @colin-rushton-qss,

the project is currently developped with Vivado 2021.1, I indeed need to make it compile with more recent versions. I'll be able to do so in 2 weeks. If you want to be able to build the design in the meantime, I would recommend either downgrading the Vivado version or try to comment out the problematic errors: These are probably false paths constraint that won't prevent a correct build if commented out, the build will just be longer/less optimal, but if passing timing, it will not be an issue.

enjoy-digital, 2025-08-12:
With https://github.com/enjoy-digital/litex_m2sdr/commit/e776e4262d174a6b10745c38aae37aa9883d3be4, the GPIOs are disabled by default.

colin-rushton-qss, 2025-08-14:
Unfortunately @1915vv-beep I'm not knowledgeable enough about this to know the fix; I'm an SWE learning this stuff recently for an SDR project I'm working on on the hobby side cause I'm a ham.  The first time I got it building was after I installed vivado 2021.2 alongside all of the 2025 litex modules.

Where are you running that from though?  It looks like an actual UI, and everything I've been building from and all the logs I get are from running the `./litex_m2sdr.py --with-pcie --variant=m2 ==build`

enjoy-digital, 2025-08-22:
Thanks for the report. This is related to a missing false path that I'll try add soon. You can already use the bitstream safely.

1915vv-beep, 2025-08-25:
@enjoy-digital Florent, tried re-building with the above mentioned commit, the setup timing issue is solved just the pulse width of the clocks: basesoc_s7pciephy_clkout0 and basesoc_s7pciephy_clkout1 are showing violation.

[image]

[image]

[vivado.log](https://github.com/user-attachments/files/21966042/vivado.log)