enjoy-digital/litex_m2sdr #54: Windows Support for LiteX M2 SDR

Ali6114, 2025-04-02:
Hi,

I’m interested in using the LiteX M2 SDR on a Windows system. I noticed that the current documentation and driver support seem to be focused on Linux.

1. Is there any ongoing work or plan to support Windows for the LiteX M2 SDR?

2. Are there any known workarounds or community efforts to make it work on Windows?

3. Would it be possible to port the existing Linux driver to Windows, or are there fundamental limitations preventing it?

Any insights or guidance would be greatly appreciated. Thanks!

enjoy-digital, 2025-04-02:
Hi @Ali6114,

for now the focus is indeed on Linux support but we are considering MacOS and Windows support. 

For the **PCIe SoC**, some interesting work has been initiated as part of Thunderscope project to have LitePCIe running on MacOs and Windows:
- MacOS: https://github.com/tech2077/litepcie-macos-driver
- Windows: https://github.com/NateMeyer/litepcie_driver_win
We are considering doing some tests with these as base and try to go further.

For the **Ethernet SoC** (when mounted in a baseboard), having MacOS and Windows support should be easier since almost everything is handled in user-space and SoapySDR driver and Ethernet is a standard interface that should not be complicated to port between the different platforms.

This work as not yet been initiated, but we could provide update here when available. Feel also free to share progress if you experiment with it.

Florent

Ali6114, 2025-04-05:
Thanks for the update, Florent!

That’s great to hear about the ongoing and planned support for MacOS and Windows, especially with LitePCIe and Ethernet SoC.

Quick question—will the LiteX Acorn baseboard mini work with the M2SDR for Ethernet-based use cases? And if so, are there any hardware or firmware considerations to keep in mind?

Also, do you have any thoughts on which Ethernet SoC configurations might be best suited for future cross-platform compatibility, especially with SoapySDR in mind?

enjoy-digital, 2025-04-05:
Hi @Ali6114,

we are indeed using the baseboard for the Ethernet SoC as you can see here: https://github.com/enjoy-digital/litex_m2sdr/issues/51#issuecomment-2765780921

The main limitation with the Ethernet Soc compared to the PCIe SoC is the available bandwidth, we currently support 1GbE or 2.5GbE on a single SFP, but we could also use the second SFP to double it if too lilmiting (and also want to see if we manage to get 5GbE working on a single SFP).

Otherwise, both PCIe and Ethernet SoC have SoapySDR support, with most of the codebase similar and the main difference in the streaming interface support.

Ali6114, 2025-04-06:
Thanks for the info!

Just to clarify—does the **PCIe SoC** support **SoapySDR** only on **Linux** for now, while the **Ethernet SoC** can work across Linux, MacOS, and **Windows**?

enjoy-digital, 2025-04-07:
@Ali6114: We only have linux support for both SoCs for now. But Ethernet SoC port on windows should require less efforts and could happen first (but still need to be planned/funded).